Azure/azure-sphere-samples #271: Couldn't send data using Private Network Service sample

nishithvpoojary, 2024-05-21:
Hi, I have been trying the Azure Sphere Private Network Service sample from Github. I have tried using both putty and Command prompt to connect to device. I was using the default eth0.  But got connection error. Please let me know how to fix this. Thank you.
[image]
[image]
[image]

jamesadevine, 2024-05-21:
Either enable eth0 or wlan0, do not have both enabled. Disable one of the network interfaces and update the sample appropriately. Presumably there is an ethernet connection between your PC and the device?

nishithvpoojary, 2024-05-22:
Thank you for your response @jamesadevine 
If i use "wlan0", application exits as shown in image.
[image]

If i use "eth0", and with wifi disabled on device, my PC coonected to same network through wifi, it didn't work.

what did u mean by ethernet connectiion between PC and device?

jamesadevine, 2024-05-22:
> what did u mean by ethernet connectiion between PC and device?

If you look at the setup section in the [README](https://github.com/Azure/azure-sphere-samples/tree/main/Samples/PrivateNetworkServices#test-the-sample), it says you can either run the app over wifi (wlan0) or ethernet (eth0). By default the application is configured to use ethernet (eth0), there will need to be a direct connection made between the device and PC (look at step 7 in the section linked above).

To test the set up, an [ethernet connection must be made](https://github.com/Azure/azure-sphere-samples/tree/main/Samples/PrivateNetworkServices#test-the-sample) between the device and PC. The commands in this section are run on a PC and communicate to the device over ethernet or wifi.

nishithvpoojary, 2024-05-22:
I have used eth0 in code and disabled wifi in device. My laptop is connected to Router through wifi and the device is connected to same router through Ethernet. Is this correct?

nishithvpoojary, 2024-05-23:
[image]

I get this error when i try with wlan0... I can use eth0 because my PC doesn't have ethernet port.

jamesadevine, 2024-05-23:
I would suspect because wlan0 has been assigned an ip address other than the one specified on the [prior line](https://github.com/Azure/azure-sphere-samples/blob/3cd45adae55eb5ac63f12f20f8e1143d1f33887c/Samples/PrivateNetworkServices/main.c#L295). Either adjust the sample to use the IP of the wlan0 interface, or set your router to assign the device the expected IP address from the sample. Otherwise, buy an ethernet to usb adapter and use ethernet!

nishithvpoojary, 2024-05-27:
I have purchased the ethernet usb adapter, but currently facing device unresponsive error for 5 devices from several days....  
[image]
I tried recovering the device... Still its not fixed.

jamesadevine, 2024-05-27:
Hiya,

Please follow this guide: https://learn.microsoft.com/en-us/azure-sphere/install/troubleshoot-installation?view=azure-sphere-integrated&pivots=windows

James

image.png (view on web)<https://github.com/Azure/azure-sphere-samples/assets/45986919/5ce13d08-dbda-4412-9a26-3a72e60134af>

You are receiving this email because you commented on the thread.

nishithvpoojary, 2024-05-28:
Hi @jamesadevine ...
The device unresponsive error was fixed after i reinstalled the Azure sphere SDK. This unresponsive issue appeared after i set the Ethernet as 192.168.100.10
[image]

I can only use the sample with 192.168.35.2
[image]
 If I change this IP to 192.168.35.3 or 192.168.35.4 in sample, it doesn't work.

jamesadevine, 2024-05-28:
Hi - do not modify the address of the ethernet "Azure sphere" ethernet adapter. This is created and managed by the Azure Sphere device communication service.

You would need to configure the IP address of the ethernet interface exposed by the usb to ethernet adapter.

Follow [this guide](https://learn.microsoft.com/en-us/azure-sphere/network/configure-ethernet?view=azure-sphere-integrated) to setup azure sphere over ethernet

nishithvpoojary, 2024-05-28:
[image]
[image]
[image]

nishithvpoojary, 2024-05-28:
The sample works now with 192.168.100.10...
There was a .exe file for my Ethernet-USB adapter that i had to run...
After that the Ethernet adapter media appeared on command prompt while running "ipconfig".

[image]

Thanks for the support @jamesadevine

closed 2024-06-07 by nishithvpoojary